kolinger/rd-usb #4: Sometimes measurements go crazy..

bursoft-git, 2019-04-23:
I measured the battery charging parameters. Everything was going fine ..
[image]
But when I turned on the screen in the tester, the readings in the WEB-GUI became absolutely wrong. At the same time on the tester screen were normal readings.
[image]
[image]
After disconnect and reconnect to the tester, the readings again became normal.
Here is the log at the time of failure indications
[image]
[image]
Such failures have happened several times in the last month, this time when the tester’s screen is turned on, and sometimes for no apparent reason. I tried different versions of rd-usb (1.0.0-beta, 1.0.0-alpha.3, 1.0.0-alpha.2), all of them are subject to this error.

Thanks for the great web GUI!

kolinger, 2019-04-23:
Application doesn't have any ability to corrupt data like this. Parser will work always wrong, not at all or always right. This changes from right to wrong on it's own. So this is what the meter actually reports over bluetooth. But why?

I did see similar issues. Few times insane values like this or more likely no values at all. When I did touch meter it did fix or brake. Turns out that my problem was in meter itself. Bluetooth module in UM34C is connected via few springy contacts (I guess this will be the same for all UM meters with bluetooth). These pads loose connection randomly. I guess because of oxidation. I did take meter apart. Scraped contacts on pcb/springy contacts little bit and then it went away.

When you press button or plug/unplug you will touch pcb and contacts inside move tiny bit and make/or brake connection with bluetooth module. These pins carry power and tx/rx. Probably TX or RX pins gets bad connection and then bluetooth board sends corrupted values away. This will explain your issues and also why you see these weird values only in GUI when screen shows correct values.  

I don't see any way how this can happen otherwise. For confirmation you can use official bluetooth app from RD and you should see exactly same values as in GUI. But since this happens randomly it takes long time to capture. In my case official app did report same nonsense as my application.

I can implement workaround for these nonsense values to drop them but then you will lose measurements. This will only hide errors. It will not solve the problem.

bursoft-git, 2019-04-23:
Indeed, the contacts may be oxidized. I'll clean them and check. Thank!

kolinger, 2019-05-07:
Did you solve it or made any progress?

bursoft-git, 2019-05-23:
I cleaned the contacts and made several measurements, the error no longer repeated

kolinger, 2019-05-23:
Great. This may help others. Seems like issue with this construction.

closed 2019-05-23 by kolinger