kolinger/rd-usb #5: Can't connect to TC66C via USB on macOS

metawops, 2019-12-21:
On my iMac (Mid 2011, no Bluetooth LE) running macOS 10.13.6 I was able to install all the requirements and launch the program with python3 (3.7.5).

The TC66C is connected via its microUSB port and a microUSB to USB-A cable to the iMac. (So not via USB-C.)

macOS enumerates the device as ```/dev/cu.usbmodemTC61``` and in Safari 13.0.4 this device is correctly listed:
[image]

Clicking on Refresh outputs this in the terminal:
```
2019-12-21 16:19:22,315 - INFO - 506bb03db78344218a77f0b686d8d25d: Received packet MESSAGE data 2["scan_serial"]
2019-12-21 16:19:22,315 - INFO - received event "scan_serial" from 506bb03db78344218a77f0b686d8d25d [/]
2019-12-21 16:19:22,315 - INFO - 127.0.0.1 - - [21/Dec/2019 16:19:22] "POST /socket.io/?EIO=3&transport=polling&t=1576941562312-38&sid=506bb03db78344218a77f0b686d8d25d HTTP/1.1" 200 -
2019-12-21 16:19:22,318 - INFO - emitting event "scan-result" to all [/]
2019-12-21 16:19:22,319 - INFO - 506bb03db78344218a77f0b686d8d25d: Sending packet MESSAGE data 2["scan-result","Results:\n<a href=\"#\" data-address=\"/dev/cu.Bluetooth-Incoming-Port\">/dev/cu.Bluetooth-Incoming-Port (n/a)</a>\n<a href=\"#\" data-address=\"/dev/cu.usbmodemTC61\">/dev/cu.usbmodemTC61 (TC66 USB Type_C Meter, manufacturer: RuiDeng, product: TC66 USB Type_C Meter, serial_number: TC66, VID_28E9, PID_018A)</a>"]
2019-12-21 16:19:22,320 - INFO - 127.0.0.1 - - [21/Dec/2019 16:19:22] "GET /socket.io/?EIO=3&transport=polling&t=1576941557979-37&sid=506bb03db78344218a77f0b686d8d25d HTTP/1.1" 200 -
```

However, when I select this device in the GUI (*not* the Bluetooth one) and try to connect no connection can be established and the status immediately goes to (stays at) "Disconnected"). 😞 
Here's the output from the terminal when I click on Connect:
```
2019-12-21 16:16:48,067 - INFO - 506bb03db78344218a77f0b686d8d25d: Received packet MESSAGE data 2["open","{\"version\":\"TC66C-USB\",\"port\":\"\",\"rate\":\"1\",\"name\":\"2019-12-21\"}"]
2019-12-21 16:16:48,067 - INFO - received event "open" from 506bb03db78344218a77f0b686d8d25d [/]
2019-12-21 16:16:48,068 - INFO - 127.0.0.1 - - [21/Dec/2019 16:16:48] "POST /socket.io/?EIO=3&transport=polling&t=1576941408064-6&sid=506bb03db78344218a77f0b686d8d25d HTTP/1.1" 200 -
2019-12-21 16:16:48,073 - INFO - emitting event "log" to all [/]
2019-12-21 16:16:48,073 - INFO - fa24272f66764704b364a156d4ac2e4b: Sending packet MESSAGE data 2["log","2019-12-21 16:16:48 - BLE address is missing. Select address in Setup\n"]
2019-12-21 16:16:48,074 - INFO - 506bb03db78344218a77f0b686d8d25d: Sending packet MESSAGE data 2["log","2019-12-21 16:16:48 - BLE address is missing. Select address in Setup\n"]
2019-12-21 16:16:48,074 - INFO - 127.0.0.1 - - [21/Dec/2019 16:16:48] "GET /socket.io/?EIO=3&transport=polling&t=1576941402197-5&sid=506bb03db78344218a77f0b686d8d25d HTTP/1.1" 200 -
```

Any ideas? 😳

kolinger, 2019-12-22:
Hi, this is bug. Bluetooth address is required for USB connection. My mistake. Update to the latest version and try again.

metawops, 2019-12-22:
> Hi, this is bug. Bluetooth address is required for USB connection. My mistake. Update to the latest version and try again.

Awesome! Works fine now! Thanks a lot for your effort!! 🙂

closed 2019-12-22 by metawops