kolinger/rd-usb #6: bluetooth UM34C linux requirements

devel0, 2020-02-10:
to make it work on linux I had to install pybluez library

```sh
python3 -m pip install pybluez
```

elsewhere I got `Attempting to use a port that is not open` error.

Could useful to report that in README, with some tips about setup for UM34C:
- pair device ( pin: 1234 )
- retrieve device address `echo paired-devices | bluetoothctl`
- connect to rfcomm `rfcomm bind 0 aa:bb:cc:dd:ee:ff`
- install python bluetooth lib `python3 -m pip install pybluez`

kolinger, 2020-02-12:
Thank you for your example. Readme updated.

This project is not using pybluez. Where did you find that pybluez fixes `Attempting to use a port that is not open`? I didn't install pybluez and your instructions did work. I did get this error when I did unplug UM34C. Maybe this is related to bad/no signal and not pybluez? After pluging UM34C back and connecting again the error was gone.

devel0, 2020-02-13:
Figured out what problem was,
ok pybluez isn't required ( I tried successfully without it )
the problem was that the device *must not paired using ubuntu Settings -> Bluetooth* because effectively it doesn't know any protocol to use and may cause interfere with serial attachment after within rfcomm and that causes that it works only sometimes.

if already paired just remove from the list of devices while address can selected and copied to clipboard:
[image]

So instructions could change within something like follow:

    - On Linux `rfcomm` can be used
        - Pair device via Settings -> Bluetooth ( Pin is 1234 )
        - Retrieve bluetooth address of paired device clicking on paired device then **remove** from paired devices or it will interfere with rfcomm
        - Bind address to serial port with `rfcomm bind 0 aa:bb:cc:dd:ee:ff`

kolinger, 2020-02-13:
Well then Settings -> Bluetooth is useless and only creates confusion. We can use `hcitool scan` instead and skip Settings. Readme updated. What do you think?

devel0, 2020-02-13:
Totally agreed, it works

```sh
devel0@tuf:~/opensource/UM34C/rd-usb$ hcitool scan
Scanning ...
	00:15:A3:00:3A:88	UM34C
```

kolinger, 2020-02-14:
Good 👍. Thanks for your cooperation.

closed 2020-02-14 by kolinger