kolinger/rd-usb #7: How to give permission

HikaruG, 2020-07-27:


kolinger, 2020-07-27:
What you mean? How to give permission to what? What OS are you using?

kolinger, 2020-07-27:
Hello, you need to give your user running rd-usb access to /dev/rfcomm0. You can do this in multiple ways. This is not specific to my application this is basic Linux behaviour.

For example - you can set chmod to be very open with `chmod 777 /dev/rfcomm0`. This is just workaround - you need to do it every reboot.
Or you can add your user to group `dialout` for example user with name "myname": `usermod -a -G dialout myname`. But this is specific to your Linux distribution. It can be different. 

Please look for guide on the internet for your specific Linux distribution. You will find many.

HikaruG, 2020-07-27:
Thanks for your reply. 
That's intriguing because I already tried adding myself to the dialout group, and it unfortunately did not change the behavior... But understood, you confirmed me that it was not on the application side, so I will look for workarounds on the internet.

closed 2020-07-27 by HikaruG

kolinger, 2020-07-27:
Yes dialout may not work if your Linux is configured in different way.

You can see what group is your Linux using for rfcomm with command `ls -l /dev/rfcomm0`. You will see something like this:
```
ls -l /dev/rfcomm0
crw-rw-rw- 1 root somegroup 5, 0 Jul 24 16:59 /dev/rfcomm0
```
In middle you gave root and somegroup. First is user and second is group. Add yourself to this group instead of dialout.