kolinger/rd-usb #8: 'You need to use the gevent-websocket server. '

v-abt, 2020-08-07:
Hey @kolinger,

I'm trying to run your web server using the latest pyhton 3.8.5 x64 but keep getting stuck on the following error message:

> 2020-08-07 09:32:38,539 - INFO - 127.0.0.1 - - [07/Aug/2020 09:32:38] "←[35m←[1mGET /socket.io/?EIO=3&transport=websocket&sid=7c385ecb3b804d909079139c1a7f26d5 HTTP/1.1←[0m" 500 -
> 2020-08-07 09:32:38,540 - ERROR - Error on request:
> Traceback (most recent call last):
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\werkzeug\serving.py", line 323, in run_wsgi
>     execute(self.server.app)
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\werkzeug\serving.py", line 312, in execute
>     application_iter = app(environ, start_response)
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\flask\app.py", line 2463, in __call__
>     return self.wsgi_app(environ, start_response)
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\engineio\middleware.py", line 60, in __call__
>     return self.engineio_app.handle_request(environ, start_response)
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\socketio\server.py", line 526, in handle_request
>     return self.eio.handle_request(environ, start_response)
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\engineio\server.py", line 362, in handle_request
>     packets = socket.handle_get_request(
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\engineio\socket.py", line 105, in handle_get_request
>     return getattr(self, '_upgrade_' + transport)(environ,
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\engineio\socket.py", line 146, in _upgrade_websocket
>     return ws(environ, start_response)
>   File "C:\Users\abt\AppData\Roaming\Python\Python38\site-packages\engineio\async_drivers\gevent.py", line 35, in __call__
>     raise RuntimeError('You need to use the gevent-websocket server. '
> RuntimeError: You need to use the gevent-websocket server. See the Deployment section of the documentation for more information.

The webserver itself is running and available via 127.0.0.1:5000 but I'm unable to communicate with my UM34C.
Is there a compatibility issue with the latest python or am I missing something else?
Your compiled app is running fine.

Thanks anyway for publishing your web server!

kolinger, 2020-08-07:
Hello,
I tried latest Python 3.8.5 in new environment. It works. I can't see how python version can cause such issue. It seems like configuration of library what I'm using for background communication is broken but I can't reproduce your setup. My app is not using `gevent-websocket`. I did make some changes to configuration of this library. Maybe this can help you. Please test latest `master`.

If problem persistst then I need more information:
Are you using python for other things? Did you use pip install for other projects?
Can you send me output of `python -m pip list` (list of your libraries)?

v-abt, 2020-08-07:
Awesome, commit 81c1aca879ae0bf4227f591340c0322126887872 fixed the issue.
Thank you for your quick reaction!

closed 2020-08-07 by v-abt