kolinger/rd-usb #9: Doesn't connect with UM25C

bravo321, 2020-09-04:
```
2020-09-04 22:37:54 - Disconnected
2020-09-04 22:45:20 - Connecting
2020-09-04 22:45:20 - Connected
2020-09-04 22:45:23 - operation failed, retrying
Traceback (most recent call last):
  File "/home/bravo123/rd-usb/webapp/backend.py", line 250, in retry
    return callback()
  File "/home/bravo123/rd-usb/interfaces/wrapper.py", line 48, in read
    return self.get_result(60)
  File "/home/bravo123/rd-usb/interfaces/wrapper.py", line 61, in get_result
    raise TimeoutError
TimeoutError
2020-09-04 22:45:23 - operation failed, retrying
Traceback (most recent call last):
  File "/home/bravo123/rd-usb/webapp/backend.py", line 250, in retry
    return callback()
  File "/home/bravo123/rd-usb/interfaces/wrapper.py", line 48, in read
    return self.get_result(60)
  File "/home/bravo123/rd-usb/interfaces/wrapper.py", line 64, in get_result
    raise ErrorException(result)
interfaces.wrapper.ErrorException: Traceback (most recent call last):
  File "/home/bravo123/rd-usb/interfaces/wrapper.py", line 100, in call
    result = callback()
  File "/home/bravo123/rd-usb/interfaces/um.py", line 37, in read
    self.send("f0")
  File "/home/bravo123/rd-usb/interfaces/um.py", line 42, in send
    self.serial.write(bytes.fromhex(value))
  File "/home/bravo123/.local/lib/python3.6/site-packages/serial/serialposix.py", line 531, in write
    raise portNotOpenError
serial.serialutil.SerialException: Attempting to use a port that is not open

```
At the same time, the other program works fine.

```
./rdserialtool  --device=um25c --serial-device=/dev/rfcomm2
rdserialtool 0.2.1
Copyright (C) 2019 Ryan Finnie

Connecting to UM25C /dev/rfcomm2
Connection established

USB: 5.048V, 0.0127A,  0.064W,  397.4Ω
Data:  0.84V(+),  2.85V(-), charging mode: Unknown / Normal
Recording (off):    0.000Ah,    0.000Wh,      0 sec at >= 0.10A
Data groups:
     0:    0.000Ah,    0.000Wh       5:    0.000Ah,    0.000Wh
    *1:    0.104Ah,    0.113Wh       6:    0.000Ah,    0.000Wh
     2:    0.000Ah,    0.000Wh       7:    0.000Ah,    0.000Wh
     3:    0.000Ah,    0.000Wh       8:    0.000Ah,    0.000Wh
     4:    0.000Ah,    0.000Wh       9:    0.000Ah,    0.000Wh
UM25C, temperature:  25C ( 77F)
Screen: 0/6, brightness: 4/5, timeout: 1 min
Collection time: 2020-09-04 23:54:16.581711
```

kolinger, 2020-09-05:
Hi, fixed in 1.5.3. Can you try current version?

bravo321, 2020-09-05:
Thanks, everything works in the latest version.

closed 2020-09-05 by bravo321