kolinger/rd-usb #10: How to get json data?

DjordjeMandic, 2020-09-07:
I wanna know how i can get json data thats shown in "Log" tab? I would like to use that to parse it in my own program and transfer it via mqtt. It could also be useful if this had mqtt support built in and url for http get requests that return last received json.

kolinger, 2020-09-08:
You can't as of now. All data is stored into local sqlite database and can be exported to CSV. There is no other way of exporting.

Implementing specific way/protocol (for example MQTT) may not be useful for other people. Other people may want different protocol or different layout for mqtt message. I would like to help you but I want some universal solution that any one can use with any protocol.

What about adding new parameter for CLI. Where you can specify program or bash script to be called when new data is received? Then you can put your mqtt code into this program/script and do whatever you want. Is this acceptable solution to your problem?

metawops, 2020-09-08:
[This](https://csvjson.com/) any help? 😳

kolinger, 2020-09-08:
MQTT is usually used for continuous/real time measurements as they happen and this can't be accomplished with current implementation. I don't believe this is about CSV vs JSON but more about how export works.

kolinger, 2020-09-20:
I'm closing this issues since no progress was done. I did implement my proposed solution - it can be used in latest version (see https://github.com/kolinger/rd-usb#custom-export) it should be good enough to workaround this problem and other similar ones.

closed 2020-09-20 by kolinger