kolinger/rd-usb #12: Firmware update

mival, 2021-01-03:
I would be great, if it is possible use this script for firmware update. Original Windows app doesn't work for me at all.

Razor512, 2021-01-03:
I just wish there was an option to customize the CSV output, e.g., automatically generating a hh:mm:ss duration output instead of just seconds, as well as a way to specify a start and end time, thus making it possible to generate a chart that accurately plots the charge time of a device. Though that will likely significantly complicate the application. 

As for firmware updates, I wonder, is the android app able to issue updates?

kolinger, 2021-01-03:
@mival Firmware update is out of scope for this application. I don't have any documentation or technical details. What I know is obtained through guessing. Experimenting with firmware is not something I want to do. It would require a significant amount of work. Even if I know how to do it I can't guarantee reliability. Wrong firmware update can easily brick devices. That's not something I want to have on my mind.

Application from manufacturer works but it's buggy. For example it's broken on some localized versions of Windows and you need to use US/English installation of Windows. I did report this to manufacturer and they know about this. Update was promised but not delivered for long time. I guess not enough people are asking for this.

@Razor512 USB connection is required for firmware upload and thus not available with Android app. For CSV related stuff - please open new issue. Customization by user is difficult and probably not needed. I would suggest adding extra column for duration of test/session as you describe. Start/end time can be extracted from first/last row so I don't see need for this. Please write you suggestion as new issue so we don't mix things together.

closed 2021-01-03 by kolinger