kolinger/rd-usb #13: Web GUI without Internet

HBerecz, 2021-08-31:
I downloaded http://127.0.0.1:5000 with WinHTTrack to D:/3/UM25C/127.0.0.1_5000 on my PC because I wold like to adapt your excellent program to my special wishes.
But the adaption of web.py from 'url = "http://127.0.0.1:%s" % port' to 'url = "D:/3/UM25C/127.0.0.1_5000"' did not work.
Could you help me to make it function? Thank you.

kolinger, 2021-08-31:
Hello, this will not work but I don't think you need this anyway.

This app doesn't use internet. It requires webserver but this webserver runs locally on your computer. You can disconnect your PC from everything and it should work just fine.

WinHTTrack is used to create snapshot of website but there is no point of doing that here. Since 127.0.0.1 is your computer. You don't need to save this URL to your HDD. It's already on your HDD!

Why you doing this? Can you please explain your goal so I can provide more helpful response?

HBerecz, 2021-09-01:
Hello, thank you for your quick response.

I would like to make some minor changes in the layout, e.g. colors of the graph and Left axis = Current, Right axis = Voltage as default.

By the way: how can I empty the Log and Data tables and force that the Graph window is cacelled and start new?

kolinger, 2021-09-01:
Then it's easiest to adjust CSS files found in `static/css` directory and JS files found `static/js` directory also HTML in `webapp/templates`. No need for HTTrack. Just explore files and you should find anything you need.

If you want to reset everything then delete data directory. On Windows you can find data directory in `C:\Users\username\AppData\Local\rd-usb` where username is name of your account. Just delete whole rd-usb directory. It will be created again on next startup. In this directory there is file `data.db` where log and data is stored. It's SQLite database and you can open this database with some database explorer program and then you can manipulate data as you want.

HBerecz, 2021-09-01:
Hello, thank you for your quick response.

I made some minor changes in the layout and colors of the Graph (Voltage = blue, Current = red according to the German conventions) in the \webapp\templates\layout.html and \static\js\application.js

kolinger, 2021-09-02:
I didn't even know there is such convention. Default colors were chosen randomly. 

More accurate and more colors can be useful for everyone. I would like to assign each unit unique color to improve distinction between units. Do you have idea what colors to use for other units? Like for power, temperature, current/power over time and resistance? I didn't find any convention.

[image]

What do you think about these colors? Blue for voltage, red for current, green for power, dark yellow for temperature, dark blue for current over time, dark green for power over time and violet for resistance.

HBerecz, 2021-09-02:
I think, this is a good idea. My favorite colors are:

Blue for voltage
red for current
green for resistance
dark yellow for power
dark blue for power over time
dark green for temperature
violet for current over time

By the way: perhaps it would be helpul for reading the Graph if the Left axis/Right axis Skale values are of the same color as the vertical descriptions and these are of the same font-size as those.

kolinger, 2021-09-02:
[image]
[image]
[image]
[image]

What you think?

HBerecz, 2021-09-03:
It's fantastic! When can I download the new version?

kolinger, 2021-09-03:
You can find it [here](https://github.com/kolinger/rd-usb/releases). So is this issue resolved then? If you have other suggestions that are beneficial to all users, let me know (as new issue).

Razor512, 2021-09-03:
Off topic, but I wanted to say thanks for making such an amazing application. 
One way that I have been using it to examine strange behaviors in some charge controllers, is to sync up values with other sources (K type thermal probes), as well as video or a charging status display.

While I am still working figuring out how to do a reliable macro for it, (trying to automate setting an offset based on the accumulated time in the skipped rows, as well as any any accumulated power related info so that there is a 0 starting point for the desired data, and then formatting the offset time values to hh:mm:ss. That then allows for easy creating of a video excerpt/ visualization of what is happening exactly when charging behavior changes. If I get a script working reliable for excel, I will try to find a place to share it :).

[image]

kolinger, 2021-09-04:
You need column where is relative time from beginning in regular hh:mm:ss format? And start counting when accumulated power is higher than 0? I have no idea how to make it in Excel but I can make it in few lines in Python. How about if such column was already in CSV?

Razor512, 2021-09-04:
That would be easier as the values could be directly used with no special macros. It helps when determining how long a device took to charge without having the few seconds of results generated when powering on the UM25C and getting the Bluetooth connected before finally connecting the load.

Usually the point I look for the load is connected which is typically indicated by the resistance not being maxed out, and then using that as the 0 point to start counting up from.

For example when I get a CSV like this, I add offsets where needed to ensure that accumulated time is starting from 0 at the moment the load is connected.
[image]

kolinger, 2021-09-04:
Accumulated time is taken from meter itself. So I didn't modify this time and added new columns named "Run time" and "Run time (seconds)" instead. And these times starts when resistance drops bellow 9999.9 Ω. This counter is based on real time (column "Time") not "Accumulated time" since TC66C doesn't report accumulated time. 

If resistance raises to 9999.9 Ω and drops bellow again timer will **not** reset. I'm not sure if it should reset or not. Most people likely won't measure multiple measurements in one session so it probably doesn't matter. Also glitch/bad contact would reset timer and that would not be desirable. So when this timer starts it will continue with same offset until end of session.

[image]

Does this solve your need?

BTW: Feel free to create another issue for different discussion next time. Hijacking other issues is not good practice.

Razor512, 2021-09-04:
That would work great. 
Sorry for hijacking the other issue, new to using the site.

kolinger, 2021-09-04:
You can try with new version. Download [here](https://github.com/kolinger/rd-usb/releases).

kolinger, 2021-09-20:
Closing this since all brought up suggestions were resolved in new version. If you have any other suggestion feel free to create another issue.

closed 2021-09-20 by kolinger