kolinger/rd-usb #14: iPad Pro version?

metawops, 2021-09-01:
I have the TC66C USB-C device and sometimes use it at my iPad Pro on its USB-C port (so not via BT LE).
It would be awesome if it was possible to rund rd-usb on iPadOS, too.
Could this work? I know rd-usb is written in Python. There are several Python implementations for i[Pad]OS, like Pythonista (rather old), Pyto and more.
What would we need to make rd-usb run on an iPad Pro?

kolinger, 2021-09-01:
I'm not sure. Most likely you can't just make couple changes and make it work. If it's even possible it would require new application for just iOS/iPadOS. Bluetooth is possible for sure but connection over USB likely not. What about running rd-usb on Raspberry PI and then connect with iPad over network?

Python by itself is not useful. More important are libraries with support for specific operating system features. For example I didn't write library for Bluetooth (pyserial/bleak) or for USB serial (pyserial) and without these libraries this application can't work.

kolinger, 2021-09-20:
I will close this now since I don't think iOS/iPadOS support is viable for this project.

closed 2021-09-20 by kolinger