kolinger/rd-usb #15: Version 1.7.1 does not seem to start (Failed to execute script app)

Razor512, 2021-10-19:
While 1.7.0 is able to start just fine, after uninstalling it and then installing version 1.7.1, it doesn't seem to start, and instead gives the following error.  

"Fatal error detected
Failed to execute script app"
[image]

Installing version 1.7.0 allows the application to start again.

kolinger, 2021-10-19:
I wasn't able to reproduce this error. I tried 1.7.1 on two different machines but both works. Thus I need your help to gather more information.

Could you run debug version of 1.7.1 and see what happens - if any more detailed errors are revealed?
https://github.com/kolinger/rd-usb/releases/download/untagged-ca6860d7ed510a97ad37/rd-usb-1.7.1-debug.zip
Here you can download zip, extract it and run "rd-usb.exe" (it is in root directory besides many files) - this doesn't need installation - and see what happens. 
If same errors occurs look for "error.log" it should be created in the same directory as "rd-usb.exe". This file should describe error in more detail.
Could you also try running this debug version via "cmd.exe"? Launch cmd, navigate to directory with "rd-usb.exe" and start "rd-usb.exe" via cmd. Then detailed error should be printed into cmd window.

kolinger, 2021-10-19:
Thank you, this seems like issue with cefpython library, not sure why it works for me. I did update integration of this library. Could you try again with this version? https://github.com/kolinger/rd-usb/releases/download/1.0.0-alpha/rd-usb-1.7.1-debug2.zip