kolinger/rd-usb #16: not an issue, how to change unite to mA ?

ens4dz, 2021-11-01:
how to change graph unite from A to milliAmps ?
thanks for this great project.

kolinger, 2021-11-01:
Hello, there is no such option. Current is available only in A. But it can be added if you think this is needed.

ens4dz, 2021-11-01:
I think it is useful to measure current for Iot modules (arduino,esp32..)

kolinger, 2021-11-01:
Alright, I did add mA as new axis option for graph. Now you can select Current (A) or Current (mA) as you wish. You can download new version [here](https://github.com/kolinger/rd-usb/releases/tag/1.8.0). Does it solve your need?

ens4dz, 2021-11-01:
[image]
 yes it's better now thanks a lot
there are some worried thing, graph didn't display correctly until press **apply** every 5s or so ! , in Amps it's display is fine.

kolinger, 2021-11-01:
I see, live update is broken and only history works. Try [1.8.1](https://github.com/kolinger/rd-usb/releases/tag/1.8.1).

ens4dz, 2021-11-01:
Awesome support, many thanks  !
it works now.

closed 2021-11-01 by ens4dz