kolinger/rd-usb #17: UM34C Windows can not add com port

ill1boy, 2021-11-12:
Hey there, 

first of all thanks for this awesome project.
I have an issue which isn't related to the project but it's prerequisites. I connected the UM34C via Bluetooth and went to bluetooth options in windows to add an outgoing com port for the UM34C. Windows tells me this device doesn't output a serial port.
Do you have any idea what I did wrong or how to fix this?

Thanks upfront :)

kolinger, 2021-11-13:
Hi, serial port should be created automatically. All what should be needed to do is to pair over bluetooth. You will not see serial port in Windows settings or in Bluetooth section but you should see serial port in Device manager. 

If pairing was successful then try to find serial/COM port in Device manager. You should find couple of them there. One of those should work.

ill1boy, 2021-11-13:
Hi, thx for the fast response. I guess my device is damaged. It even can't connect to the Android app. There are no serial ports registered when pairing. :(

Razor512, 2021-11-13:
If the device is not functioning properly, make sure the Bluetooth module is making proper contact main PCB of the  UM34c, as the Bluetooth board physically presses against 4 pads on the main PCB and a spotty connection there will cause issues.

Beyond that, make sure there are no cold solder joints on the Bluetooth board.

kolinger, 2021-11-14:
Seems like something is wrong with your device but Bluetooth can be funny sometimes. Did you try to remove pairing and do it again? Do you have up to date firmware? Maybe firmware upgrade via official app could help?

fragsalat, 2021-11-19:
Awesome. Releasing the screws and bending the pins a bit solved the issue and I'm able to connect via phone and your app 👍 
Still I had to manually create the serial port but now it's working. Thanks for your support and your awesome work <3

ill1boy, 2021-11-19:
Awesome. Releasing the screws and bending the pins a bit solved the issue and I'm able to connect via phone and your app 👍Still I had to manually create the serial port but now it's working. Thanks for your support and your awesome work <3

closed 2021-11-19 by ill1boy