kolinger/rd-usb #18: win8 32 bit problem

Uwe-B, 2021-12-09:
Hi, this is just awesome if I could get it to work! 
I have a win 8 (32bit) laptop and tried to get it running on it (couldn't use the exe files though since they are for 64bit machines). 
- Python (3.8) installed properly, 
- pip install -r requirements.txt run through without error 
but while trying to run the webserver (python web.py) I got an error regarding the wrong windows version:
`d:\Utilities\USB\RD-USB\rd-usb-master>python web.py
Traceback (most recent call last):
  File "web.py", line 19, in <module>
    from webapp.backend import Backend
  File "d:\Utilities\USB\RD-USB\rd-usb-master\webapp\backend.py", line 16, in <module>
    from interfaces.tc import TcBleInterface
  File "d:\Utilities\USB\RD-USB\rd-usb-master\interfaces\tc.py", line 6, in <module>
    from bleak import BleakClient
  File "C:\Users\U\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\__init__.py", line 70, in <module>
    raise BleakError(bleak.exc.BleakError: Only Windows 10 is supported. Detected was ('8.1', '6.3.9600', 'SP0', 'Multiprocessor Free')`

kolinger, 2021-12-09:
Hello, seems like the Bluetooth BLE library doesn't support Windows 8, only Windows 10. This can't be fixed. Only Windows 10 and newer supports Bluetooth BLE. But you may not need BLE.

What RD USB meter are you using? BLE is required only for TC66C over Bluetooth. UMx and TC66C over USB should work on Windows 8.

I did make change so TC66C BLE support is optional. This should enable to launch the application even on older systems. Of course without TC66C BLE support available. You can try latest source code from repository.

Uwe-B, 2021-12-09:
Thanks. I have the UM34C, will try it, just figured out that my other Laptop (Win7 64bit) does recognize my UM34C though it doesn't have BLE.

kolinger, 2021-12-09:
Then you should be fine with updated code. UM34C uses regular bluetooth so it should work anywhere. Well in theory. I don't think anyone tested this on Windows 7 or 8 yet, you may be the first!

Uwe-B, 2021-12-09:
OK, win 7 install will not work (pip install -r requirements.txt) :
`Collecting pendulum~=2.1.2
  Using cached pendulum-2.1.2-cp37-cp37m-win_amd64.whl (129 kB)
Collecting pyserial~=3.5
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting python-socketio~=5.4.0
  Using cached python_socketio-5.4.1-py3-none-any.whl (55 kB)
Collecting python-engineio~=4.2.1
  Using cached python_engineio-4.2.1-py3-none-any.whl (52 kB)
Collecting appdirs~=1.4.4
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting bleak~=0.12.1
  Using cached bleak-0.12.1-py2.py3-none-any.whl (123 kB)
Collecting pycryptodome~=3.9.0
  Downloading pycryptodome-3.9.9-cp37-cp37m-win_amd64.whl (14.1 MB)
     |██████                          | 2.5 MB 2.2 MB/s eta 0:00:06ERROR: Could
not install packages due to an OSError: [WinError 8] Für diesen Befehl ist nicht
 genügend Speicher verfügbar
--- Logging error ---
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-
8'>`

I'll try it on the other device.

Uwe-B, 2021-12-09:
Did the test on the other (Win8 32bit) device, same? error as before:
`d:\Utilities\USB\RD-USB\rd-usb-master>python.exe web.py
Traceback (most recent call last):
  File "d:\Utilities\USB\RD-USB\rd-usb-master\interfaces\tc.py", line 8, in <module>
    from bleak import BleakClient, BleakError
  File "C:\Users\U\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\__init__.py", line 70, in <module>
    raise BleakError(bleak.exc.BleakError: Only Windows 10 is supported. Detected was ('8.1', '6.3.9600', 'SP0', 'Multiprocessor Free')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "web.py", line 19, in <module>
    from webapp.backend import Backend
  File "d:\Utilities\USB\RD-USB\rd-usb-master\webapp\backend.py", line 16, in <module>
    from interfaces.tc import TcBleInterface
  File "d:\Utilities\USB\RD-USB\rd-usb-master\interfaces\tc.py", line 11, in <module>
    except BleakError as e:NameError: name 'BleakError' is not defined`

kolinger, 2021-12-09:
Yes, indeed there was mistake in my fix, fixed now. Tested on Win7 32bit and now application will launch.

First computer doesn't have enough memory? Seems like generic operating system error than error in application.

Uwe-B, 2021-12-09:
Thanks a lot, got it working on main machine (Win7 64bit). 
Error above was due to trying to install python dependencies with root privileges. I will test it on win 8 laptop too (have to charge it before), just to inform you of the results but I'm very glad that the win7 (64bit) machine is working with it.

Uwe-B, 2021-12-09:
Successfully tested the patch on the other machine too (win 8.1 32bit).
Thank you very much for your efforts, I was afraid I had to use the chinese Android app, this is much better.

closed 2021-12-09 by Uwe-B