kolinger/rd-usb #20: Feature Request: UI - Dark Mode

dmberrycom, 2022-01-06:
Add an option to switch UI from a White to Black theme to save my eyes :)

kolinger, 2022-01-08:
Hello,
dark theme was added. You can find it in Setup tab. Available in 1.12.0 version - [here](https://github.com/kolinger/rd-usb/releases).

It is sufficient?

dmberrycom, 2022-01-08:
Thanks, I think there a few issues with the dark bootstrap theme.

I started a fork to look at adding an additional theme and a few changes. I used the dark theme as the baseline but it seems that the source for the dark theme is not quite the same as the default(light) theme. 

The dark theme has issues with the navbar expanding and hiding parts of the UI and also the graphs are only using small proportion of the screen height in dark mode.

kolinger, 2022-01-09:
Hi,
issues are expected but it shouldn't be hard to fix.

Navigation issue: 
I did tweak the navigation a bit - now it fits so it doesn't expand when not needed. Also even if it does expand it will not hide content as before. This should fix the navigation issue.

Graph height issue:
I wasn't able to reproduce this issue. In my testing graph height is acceptable and identical in light and dark theme.
Here I need your help. If I can't see the problem then I can't fix it, but you can give me more information, then I could perhaps reproduce this issue on my end or at least I could understand the issue more. Can you provide screenshot? Do you use CLI+external web browser or standalone version of rd-usb?

dmberrycom, 2022-01-09:
Great, 

Absolutely happy to provide more information would you like to push the update with the navigation fix, I will then re-test against that.

So far I've just been doing the testing in CLI+external web browser, but I could also try it with the standalone.

kolinger, 2022-01-09:
Hi,
you can get temporary build with fixed navigation [here](https://github.com/kolinger/rd-usb/releases/download/1.0.0-alpha/rd-usb-navigation-fix.exe).

kolinger, 2022-01-09:
New themes and fixes available in 1.13.0 - [here](https://github.com/kolinger/rd-usb/releases).

dmberrycom, 2022-01-09:
Super,

Thanks for such a handy tool.

closed 2022-01-09 by dmberrycom