kolinger/rd-usb #22: Version 1.12.0 keeps disconnecting from TC66C (USB) with IndexError on macOS

metawops, 2022-01-09:
I just after a long time tried to get this to work on macOS again and wasn't very successful.
After getting the latest version (1.12.0) I did the `pip3 install -r requirements.txt` which worked fine (despite the "DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future." warnings).
Launching with `python3 web.py 5555` (port 5000 was already in use) worked fine, too.
Listing the via USB connected TC66C worked, too (it's called `/dev/cu.PT-P710BT7085` here).
Connecting to it worked, too, at first. But no data could be shown.
After a few seconds the log shows this over and over again every 10 seconds:
```
2022-01-09 13:48:13 - operation failed, retrying
Traceback (most recent call last):
  File "/Users/stefan/rd-usb/webapp/backend.py", line 295, in retry
    return callback()
  File "/Users/stefan/rd-usb/interfaces/wrapper.py", line 48, in read
    return self.get_result(60)
  File "/Users/stefan/rd-usb/interfaces/wrapper.py", line 64, in get_result
    raise ErrorException(result)
interfaces.wrapper.ErrorException: Traceback (most recent call last):
  File "/Users/stefan/rd-usb/interfaces/wrapper.py", line 101, in call
    result = callback()
  File "/Users/stefan/rd-usb/interfaces/tc.py", line 151, in read
    return self.response.decode()
  File "/Users/stefan/rd-usb/interfaces/tc.py", line 228, in decode
    if self.decode_integer(data, 88) == 1:
  File "/Users/stefan/rd-usb/interfaces/tc.py", line 250, in decode_integer
    temp4 = data[first_byte] & 255
IndexError: index out of range
```

I don't know if you can do anything about it or if it helps. I know that macOS compatibility is not your thing but for me it would be great if it worked. Let me know if I can help out / try things / further debug anything for you. I'm running Python 3.9.9 on macOS 12.1 (Monterey).

kolinger, 2022-01-09:
Hi,
could you provide more of log what happened before this? This error should never happen if communication between PC and TC66C is working. Are you sure that /dev/cu.PT-P710BT7085 is really TC66C? This looks like communication with TC66C is not working and /dev/cu.PT-P710BT7085 is responding but with unknown data format - thus most likely explanation is that this is not TC66C but another device.

metawops, 2022-01-09:
> Hi,
> could you provide more of log what happened before this? This error should never happen if communication between PC and TC66C is working. Are you sure that /dev/cu.PT-P710BT7085 is really TC66C? [...]

Okay, I will further investigate. Stay tuned! 😉

metawops, 2022-01-09:
Fascinating: the TC66C doesn't get recognized by macOS on my Intel i9 iMac from 2020 when connected via its male USB-C plug directly to the iMac.
The only way to make the TC66C appear as a `/dev/tty*` device is when using its Micro-USB port with a Micro-USB to USB-A cable at a normal USB-A port on the iMac.
So having a USB-C port equipped data logger is totally useless here.
So then I wanted to switch back to my UM34C but when I connect this to my iMac via the Micro-USB port (apparently only this supports USB-HID?) no new serial device appears in `/dev` either. – Which serial chipset is used inside the UM34C? I do have the latest FTDI driver for macOS installed ... But maybe it's not an FTDI chip inside the UM34C?

kolinger, 2022-01-09:
USB C ports on TC66C are for measuring only. Data connection is available only with micro USB port. This is nothing related to your system. This is just how TC66C is designed. You plug one USB C into power source, second USB C to your load and micro USB into PC for logging.

UM34C for change doesn't support USB data communication at all. Only Bluetooth is supported. If you want to use USB then you can only use TC66C via micro USB. Everything else is Bluetooth based. Well as far as I know. At least this is what rd-usb is supporting.

metawops, 2022-01-09:
I see. Thanks a lot for clarifying this for me! 👍👍

closed 2022-01-09 by metawops