kolinger/rd-usb #23: Web-Interface: currently running version?

metawops, 2022-01-09:
More of a suggestion: I'd love to be able to see what version I'm running but I can't find that information in the web GUI.
The browser title currently (1.13.0) contains just "RD-USB v0". Maybe at least this can be upgraded to something like "RD-USB v1.13.0"? Or add a little label in the new Settings tab that just tells the currently running version. 😉

kolinger, 2022-01-09:
If you download source code with GIT then you will get version. Otherwise application doesn't know. This is intentional. The only way how to display version without GIT is to hardcode version into source codes - then I would need to remember always change version in source code. This is not something I want to deal with all the time.

Install git (if you don't have it already) and then download source code with `https://github.com/kolinger/rd-usb.git`. Then version should appears if `git` command is available in terminal.

metawops, 2022-01-09:
Oh, *I* know which version I'm running. I just thought about any user who will use the web GUI and who's not the one who downloaded or git-pulled the program. 🤷‍♂️

kolinger, 2022-01-09:
Well like 99% of people will have version. This is not about web GUI but how you obtain source code.
Most people use binaries for Windows - there you always have version, including web GUI.
Some people use Linux, then they likely use GIT - they will have version also, including web GUI.
Very few people will obtain source code outside GIT and run python manually - then there is no version. This is what we talking about here. But that's very few people. Most people are covered.

kolinger, 2022-01-09:
Well I found alternative that will not add more work me.

Now you can download prepared source code from [release page](https://github.com/kolinger/rd-usb/releases). `rd-usb-source-x.zip` is source code with version. If you start with python this source code then you will have version also. Now all people can have version no matter how they use this application. This should help people if they don't have Windows and don't have GIT and do know how to use python.

Do you think this solution is acceptable?

metawops, 2022-01-09:
Sure. Thanks. It's just a very very tiny little thing ... 😉 (I was thinking of MakerSpace with lots of PCs running Linux and rd-usb for visitors of the MakerSpace who haven't installed it by themselves ...)

closed 2022-01-09 by metawops