kolinger/rd-usb #24: Data column widths changing in graph view.

dmberrycom, 2022-01-09:
Just a slight readability/rendering issue in the data & graph values on screen.

On the graph menu the measured values can have a different number of decimal places displayed for each measuring point. This causes column widths to dynamically change on screen when data is refreshed or changes.

It's most prevelent on the voltage, current and data +/- fields.

On the data menu, this is not so much of a big concern but it would be good to have all numbers displayed to the same number of decimal places to improve readability.

kolinger, 2022-01-10:
Hi,
movement of columns can't be completely prevented - for example when Voltage column changes from 1V to 10V it will shift other columns. Similar for Mode column where many names with different lengths are possible. This has no real applicable solution. This is perhaps not what you mean? You are focused only on decimal places?

I will try to make number of decimal places fixed everywhere. It's not simple since UM24/UM34, UM25 and TC66 all have different number of decimal places, and different kind of values have different number of decimal places for each kind of meter too. Even more complicated is data table where I don't know what data was sampled by what kind of meter and each meter has wildly different number of decimal places (UM24/UM34 has only 2 decimal places for Voltage yet TC66C has 4). Showing less decimal places than meters supports is bad. Showing more is also not ideal.

Very simple thing yet quite complicated!

closed 2022-01-10 by kolinger

kolinger, 2022-01-10:
Fixed number of decimal places are available in 1.14.0-beta - [here](https://github.com/kolinger/rd-usb/releases).

dmberrycom, 2022-01-11:
Yes, it always turns out that what looks simple on paper always turns out to be much more complex.

I have tested today and your solution works well.

Thank you

commit 5171d89e (2022-01-10): sessions overhaul, also added rounding to fixed number of decimal places
this fixes #24, also fixes "current" issue from #19, this allows to create different sessions/measurements with identical name