kolinger/rd-usb #25: Application keeps hanging on "Loading..."

SgtBlade, 2022-02-09:
I've tried 1.13 and 1.14 Beta but both are stuck on the "Loading...". 
I tried using both the installer and the lightweight version.

The 127.0.0.1:5000 gives Error 500: 
Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

I'm on Windows 11 so it might not be compatible yet? Are there any log files I can check?

Console of the lightweight version gives this error:
  ```
  File "C:\Users\******\AppData\Local\Temp\_MEI217802\webapp\templates\layout.html", line 12, in top-level template code
    <link href="{{ url_for('static', filename='css/themes/' + theme + '.bootstrap.min.css') }}" rel="stylesheet" />
jinja2.exceptions.UndefinedError: 'theme' is undefined
2022-02-09 14:18:35,766 - INFO - 127.0.0.1 - - [09/Feb/2022 14:18:35] "←[35m←[1mGET / HTTP/1.1←[0m" 500 -
2022-02-09 14:18:36,286 - ERROR - Exception on / [GET]
Traceback (most recent call last):
```

kolinger, 2022-02-09:
Not Windows 11 related. This was bug affecting all new installations since 1.13.0 on all systems.

Fixed in 1.14.1 - available [here](https://github.com/kolinger/rd-usb/releases).

Please let me know if fix works for you too.

SgtBlade, 2022-02-09:
Jup, that fixed it.

There is another minor bug I found when moving the window to my second monitor:
[image]

Though this is something minor as it works perfectly well on my main monitor

kolinger, 2022-02-09:
Window in a window? Very strange. What happens if you click on COM4 or some navigation item? When does this happen exactly? If you move opened window from one to another screen or does this happen on opening of application? I can't reproduce your result.

kolinger, 2022-02-09:
Looks like rendering issues. Perhaps related to accelerated rendering via GPU. You can try to disable GPU rendering as workaround. Modify your shortcut to rd-usb by appending "--disable-gpu". You need new version for this - 1.15.0 - available [here](https://github.com/kolinger/rd-usb/releases).

Like this:
[image]

SgtBlade, 2022-02-09:
Disabling the GPU fixed it, thanks!

closed 2022-02-09 by SgtBlade