kolinger/rd-usb #30: Example service?

Bluscream, 2022-03-29:
Since the readme states ``

i would like to create a service that ensures proper operation after a reboot;
```
[Service]
ExecStart=python3 /srv/usbmeter/rd-usb/web.py --daemon --prefix /usb/ # Start daemon
Restart=on-success
RestartSec=1m
User=www-data

[Install]
WantedBy=multi-user.target

[Unit]
Description=USB Meter Monitoring Webserver
After=network.target
```

this is what i have to far but sadly it does not seem to be able to start everything; especially the bluetooth connection and serial binding. Could you provide some help with this?

kolinger, 2022-03-29:
You need to create another service pro rfcomm since it needs to be run as root.

For example /etc/systemd/system/rfcomm0.service could look like:
```
[Unit]
Description=rfcomm0
After=bluetooth.target

[Service]
Type=oneshot
ExecStart=/usr/bin/rfcomm bind 0 BLUETOOTH_ADDRESS_HERE
ExecStart=/usr/bin/chown YOUR_USER: /dev/rfcomm0
ExecStop=/usr/bin/rfcomm release 0
RemainAfterExit=yes
Restart=no

[Install]
WantedBy=multi-user.target
```

And of course `systemctl enable rfcomm0.service`

Bluscream, 2022-03-29:
Thanks a lot :)

closed 2022-03-29 by Bluscream