kolinger/rd-usb #34: Timeout Error

NastyFlytrap, 2022-08-07:
2022-08-07 05:07:56 - Connected
2022-08-07 05:08:06 - operation failed, retrying
Traceback (most recent call last):
  File "webapp\backend.py", line 297, in retry
  File "interfaces\wrapper.py", line 48, in read
  File "interfaces\wrapper.py", line 61, in get_result
TimeoutError

My UM34C keeps blinking blue, which usually means ready to pair. WebGUI says its connected while the meter is still waiting for a connection, and in the GUI it errors out, despite saying its connected

kolinger, 2022-08-07:
Connected just means that serial COM port is open. This doesn't mean communication is working.

Timeout means application didn't get expected response. Application first asks for data and then waits for result. This is where timeout expires. Since this is first response then communication is likely not working at all. 

Make sure UM34 is paired and you have correct serial COM port selected. If you use other available and unused serial COM where UM34 isn't then you will get exactly same response. Since unused serial COM port can be opened yet there will never be any valid response since there is no UM34 on the other end.

UM34 blinks when there is no active connection. Light should be solid or at least blinking pattern should be interrupted when application is trying to communicate. If light blinks all the time same then application doesn't reach UM34 at all and serial COM port is likely wrong.

NastyFlytrap, 2022-08-08:
Ah, right. I tried pairing it and i couldnt find it, but today it decided to show up under the first bluetooth devices menu, the one with the pens, keyboards and mice. 1234 is the pin, it paired, and i selected COM3 and now it works, thanks!

Sorry for wasting your time

closed 2022-08-08 by NastyFlytrap