kolinger/rd-usb #35: Would it be possible to set a different location for the data and configs and such?

NastyFlytrap, 2022-08-11:
Appdata/local is nice and all, but i'd like to store everything thats there in a different folder. Could you add an option to specify that location in the WebGUI?

kolinger, 2022-08-11:
Hi,
application needs to know location before it starts thus setting location in web UI is too late.

What about if you could specify location with command-line argument via shortcut for example?

NastyFlytrap, 2022-08-11:
> Hi, application needs to know location before it starts thus setting location in web UI is too late.
> 
> What about if you could specify location with command-line argument via shortcut for example?

Its fine by me, i know my way around a terminal, but a good amount of people dont. Maybe the ability to set it in the config file? 

I imagine it could just read where the config file tells it to save, and move all of its files there, save the last location it copied to, and on next bootup it'd compare the location in its memory to the config file, and if they match, skip the copying.

Or maybe thats too convoluted. Im not sure what the most user-friendly solution is.

kolinger, 2022-08-11:
This is not feasible. There is no way to modify exe by itself and even to modify other exe is very complicated since it's in compressed form and not directly readable. This could be done only if you save another file besides exe in directory where is exe located. Do you think this is cleaner solution than CLI option?

NastyFlytrap, 2022-08-12:
> This is not feasible. There is no way to modify exe by itself and even to modify other exe is very complicated since it's in compressed form and not directly readable. This could be done only if you [...]

Not really, but my goal was user accessibility, it'd deffinitely be a lot more pain and work for us, who have to think this up, create it, and bugfix it.

kolinger, 2022-08-12:
I'm afraid there is no user-friendly way how to do this properly. At least there is some way how to do it now - see [readme](https://github.com/kolinger/rd-usb#custom-data-directory) for details. Available in 1.7.0 version.

Does this solve your issue?

NastyFlytrap, 2022-08-12:
> I'm afraid there is no user-friendly way how to do this properly. At least there is some way how to do it now - see [readme](https://github.com/kolinger/rd-usb#custom-data-directory) for details. Available in 1.7.2 version.
> 
> Does this solve your issue?

It sure does, thanks

closed 2022-08-12 by NastyFlytrap

NastyFlytrap, 2022-08-12:
Wait, how do i set the option? I cant put that on an exe file because windows thinks i want to rename the file, and if i were to put that on a shortcut then it wouldnt look like your example, then it'd include its full location, and it'd be in quotation marks aswell...

kolinger, 2022-08-12:
You can make shortcut. If you have spaces in path then you need to use quatation marks but only for paths not for option itself, like this:

`"C:\some\program.exe" --data-dir "C:\data"`

NastyFlytrap, 2022-08-12:
Yea, thats kinda why i asked, since the formating in your example on the main page doesnt look like the text i'd find in a shortcut, which was kinda mislead. Thanks