kolinger/rd-usb #36: Would it be possible to set the connection timer to a very long time? so it would reconnect automatically

custom46, 2022-08-11:
I'm using your gui to record data from a solar panel, with which I can monitor solar radiation. The thing is, sometimes the clouds reduce the power output of the panel to barely zero and the UM25C shuts off. I want to reccord all day, so when the sky clears up, it would be nice to reconnect automatically to the device and continue reccording. Thanks!

kolinger, 2022-08-12:
Hi,
I need little bit more information to know exactly what would help you. 

For example I can think of two possible situations:

1) UM25C and machine with rd-usb shares power and thus when UM shuts of rd-usb application also shuts off. As is you need to click Connect each time manually. Solution could be to make configurable setting to enable auto connect on startup.

2) Machine with rd-usb has constant power and only UM loses power. As is rd-usb will reconnect/retry automatically but only fixed number of times, if too many errors occur after each other then rd-usb gives up. Solution could be a setting to disable giving up (=retry forever).

You are experience first/second/both or completely different situation?

custom46, 2022-08-12:
Hi,

Thank you for taking time to answer me.

I am in the second situation you described. A pc will be running on a powersource and will stay on, but the UM will only be connected to a small solar panel out in the field. 

I tought about plugging a powerbank to the micro-usb port to power the UM when solar is not enough but it affects the readings considerably. The other solution, as you mentioned, would be to wait longer for the UM to reconnect.

kolinger, 2022-08-12:
I added new options in Setup tab. Where you can set Timeout and Retry on error to very high value or to 0 to completely disable giving up and thus indefinitely retrying connection - this way rd-usb should reconnect when USB meter is available again even after long periods. Available in [1.7.0 version](https://github.com/kolinger/rd-usb/releases/tag/1.17.0).

This is for sure easier to solves in software than trying to use power bank.

Let me know if these options work for you.

custom46, 2022-08-12:
Thank you for the fast update!

I tested every options, and I can say that the "retry on error" does not increse the number of tries (low timeout).

It works with setting the timeout to zero and it fits my use case nicely! The only thing is that while searching (trying to open the COM port), the cpu usage skyrockets. I see that this operation is ongoing for about 5 seconds and then it retries. Would it be possible to set a time interval between the searches? My reading rate is very long (10s is the max, more would be nice), so I don't mind if it doesn't reconnect in the minute.

I hope i'm not asking too much of you!

Thanks again

kolinger, 2022-08-12:
Number of retries and timeout work together. If error occurs then both number of retries and timeout can terminate connection but only one will depending on what error is occurring.

We need to have both since there can be very quick errors occurring fast after each other. In such case number of retries will terminate connection before timeout can. Other kinds of errors can take long time like one minute per try. In such case timeout will terminate connection before number of retries can. Point of is to make application more responsive so we need to have two metric for terminating connection. 

About the CPU usage - there was bug where background process used 100% of one CPU core in some situations and unnecessary high amount of CPU in rest of the situations, this is fixed in version 1.7.2.

Several longer sampling periods were also added.

Let me know if CPU usage is now better.

custom46, 2022-08-12:
I understand, setting everything to zero works for me.

The CPU usage definitely dropped, very low usage now.

All in all, everything works! Thank you very much for all the help. My use case is to monitor power out of a solar panel to measure solar radiation. It is part of a data gathering study in Tanzania, witch which we will design water supply solutions.

I hope this was not too much trouble for you! Thank you again

closed 2022-08-12 by custom46