kolinger/rd-usb #37: Rendering issue for current (A), current (mA),and power (W) when graphing for multiple hours. All other values display properly

Razor512, 2022-10-13:
I noticed a weird issue when logging for a longer period of time recently. After a while, the the current and power related values will stop displaying, though all other values will display just fine in the graph. Furthermore, restarting the app and loading the session does not resolve the issue. 

I also recorded a video of the issue. The issue happens regardless of which axis is chosen to display the value. This issue only happens when collecting data for multiple hours; in my case, around 14 hours.

https://user-images.githubusercontent.com/75226214/195699069-f8708ff9-6947-44fd-bc4d-4277b0dbbee7.mp4

[2022-07-20 2 issue.csv](https://github.com/kolinger/rd-usb/files/9780291/2022-07-20.2.issue.csv)

kolinger, 2022-10-14:
Strange indeed. Seems like issue with amCharts 4 library.

I upgraded to next generation of amCharts 5 and now graph works fine even with your dataset. amCharts 5 is very different internally and should also work faster/more efficiently with bigger datasets. This change forced me to do total rewrite of graph configuration - this may cause new issues.

Let me know if [1.18.0](https://github.com/kolinger/rd-usb/releases/tag/1.18.0) helps and doesn't introduce new issues.

kolinger, 2023-05-27:
Closing this due to inactivity and this issue should be fixed anyway. If issue still persists let me know.

closed 2023-05-27 by kolinger