kolinger/rd-usb #42: Interface needs a way to reliably reset accumulated power between sessions (or to choose not to?)

SiteRelEnby, 2024-03-06:
Sometimes, after recording a session, when I go to start another the accumulated power doesn't reset.

A couple of possible solutions:
* Always force accumulated power to 0 when starting logging
* A way to reset accumulated power, like a webUI button
* Some kind of setting to reset it or not when starting logging

Great project, very useful :)

kolinger, 2024-03-06:
This app reflects numbers on the display of the USB meter more-or-less and that's why it's possible to do new session and still have previous accumulated data since they didn't reset on the USB meter itself and this apps just reads what meter reports. 

That's intendent way of reset of accumulated data is via the USB meter interface. UMxxC clears the accumulated data by holding PREV button. UCxxC clears the data by holding K1 button. You don't like this?

I don't know if these meters support reset of accumulated data remotely via some command. Quick research shows this is likely possible for UM but not for TC. I don't like to have something for one type and not the other type, that will just make people ask why it's missing for the other. That's why reset via the meter itself it's better since it's universal.

There is always software way - we could just start measurement with current accumulated values and then it's easy to substract this starting value from every sample and thus you will get relative numbers starting from 0 but this way the meter will still show the continuous accumulated values though so only RD-USB will show the "correct" values.

Razor512, 2024-05-06:
One method I use is with an old smartphone running the old Ruideng android app, and I will simply disconnect the rd-usb app from the device, then have the Ruideng app connect to the UM25C, and that will allow for the app to reset the values, then I have the rd-usb application reconnect.

kolinger, 2024-07-15:
I added ability to zero accumulated values in table view and also added zeroed values to CSV export. It's simple software side subtraction but it should allow you to see accumulated mAh or mWh value after measurement is done even if you did forget to reset the meter.

I don't think there is option to reset the accumulated values on the meter for TC66C, that's why I added software side solution that will work for all meters.

closed 2024-07-15 by kolinger