kolinger/rd-usb #43: Accessing data files from the meter.

nath183, 2024-03-08:
Hi,

I am trying to use the rd-usb project to be able to retrieve the raw data from the UM25C power meter so that I can use the readings within another project, my aim is to launch the rd-usb program from within another Python program that I am developing and then I can utilise the data which it produces.
I have seen that it populates a SQLite database if you specify a directory when launching it from the command line however I am struggling to understand how to get it to consistently generate JSON files containing the power readings, ideally I would like 1 file with all the readings in since starting the program or a file containing a single reading every x amount of seconds.

Currently I have been launching it with the following command:
python web.py --data-dir "C:\rd-output" --daemon --on-receive on-receive.cmd
However only sometimes does it generate a JSON file and the file names are usually like this "on-receive-payload-1709934052.004884.json", is there anyway to alter the filename so it is easier to pickup from another program.

Any help would be much appreciated.

Thanks,
Nathan

kolinger, 2024-03-09:
Hi,
the --on-receive is the simplest way to get json. Replace on-receive.cmd with your own scripts (perhaps written in python as well) and then it's very easy to process the data, you will receive the file as first command line argument, thus you don't care what si the name and you can delete it after processing. The frequency can be specified with --on-receive-interval see [readme](https://github.com/kolinger/rd-usb?tab=readme-ov-file#custom-export) for details.

I'm not sure if you saying that --on-receive wasn't called in specified interval (default 60s) or you didn't know there is this interval? There is a option that --on-receive won't be called if there are no new data otherwise it's called as per --on-receive-interval, well it should be.

nath183, 2024-03-09:
Hi,
Thanks for your response I think I understand but I'm struggling to make it work.
I created a simple script which should either print out the file name or print out the data whichever is passed to my script.
`# json_writer.py

import sys
import json

print("## JSON WRITER OPENED ##")
print(f"NUM ARGUMENTS: {len(sys.argv)-1}")
print(f"FIRST ARGUMENT: {sys.argv[1]}")`

And I am starting the rd-usb software with the following command
`python web.py --data-dir "C:\rd-output" --daemon --on-receive json_writer.py`

However it just gives me this error
`2024-03-09 23:43:00,953 - ERROR - <class 'OSError'>
Traceback (most recent call last):
  File "C:\Users\Nathan\OneDrive - Newcastle University\Stage 3\CSC3094 - Dissertation\Additional files for code\rd-usb-1.22.0\webapp\backend.py", line 273, in run
    self.update(data, version)
  File "C:\Users\Nathan\OneDrive - Newcastle University\Stage 3\CSC3094 - Dissertation\Additional files for code\rd-usb-1.22.0\webapp\backend.py", line 336, in update
    subprocess.Popen(command, shell=True, env={})
  File "C:\Users\Nathan\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Nathan\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] The parameter is incorrect`

kolinger, 2024-03-10:
You passing .py script directly but --on-receive is trying execute valid program/command, .py script isn't valid program/command.

I updated rd-usb for better script compatibility and added python example - this should get you started right away.
Use 1.22.1 version where you can also find the new example.

nath183, 2024-03-11:
Hi,

I understand how it works now.
Thanks so much for the quick response and python example.

closed 2024-03-11 by nath183