kolinger/rd-usb #45: Help needed: UM25C connection fails with "Bluetooth service not found, try to initiate Setup again"

AlexHuebi, 2024-05-05:
Hi there. 

I tried to use the rd-usb to better utilize my um25c usb meter. 
I bought the meter [from here](https://de.aliexpress.com/item/32855845265.html), which as far as I could tell, should be the "official" Storefront. 

I am also using the UM Meter App on my phone which works, but just barely. (plagued by bad UI)
That also brings be back to this repo.  

I tried to use rd-usb on multiple Windows (one has the manufacturing date at 2024 and one 2017) Laptops, on my Linux Machine and on an Raspberry Pi 4. 
While rd-usb finds the meter just fine, it always throws the error "interfaces.interface.FatalErrorException: Bluetooth service not found, try to initiate Setup again". 
Which I traced back to the "um.py" File on line 100 where bluetooth.find_service is called, but returns an empty list. 
The address is set correctly but I lack the knowledge to further debug anything that has to do with Bluetooth so any help, that would get this back up and running would be very appreciated.

Full Stacktrace: 
```
Traceback (most recent call last):
  File "/root/rd-usb/webapp/backend.py", line 252, in run
    self.retry(self.interface.connect)
  File "/root/rd-usb/webapp/backend.py", line 363, in retry
    return callback()
           ^^^^^^^^^^
  File "/root/rd-usb/interfaces/wrapper.py", line 34, in connect
    self.get_result(60)
  File "/root/rd-usb/interfaces/wrapper.py", line 65, in get_result
    raise FatalErrorException(result)
interfaces.interface.FatalErrorException: Traceback (most recent call last):
  File "/root/rd-usb/interfaces/wrapper.py", line 112, in call
    result = callback()
             ^^^^^^^^^^
  File "/root/rd-usb/interfaces/um.py", line 108, in connect
    raise FatalErrorException("Bluetooth service not found, try to initiate Setup again")
interfaces.interface.FatalErrorException: Bluetooth service not found, try to initiate Setup again
```

One thing that I have noticed that may help: I am unable to connect via the normal bluetoothctl to the USB Meter: 
```
[bluetooth]# connect 00:15:A3:00:5B:49
Attempting to connect to 00:15:A3:00:5B:49
[bluetooth]# hci0 00:15:A3:00:5B:49 type BR/EDR connected eir_len 12
[CHG] Device 00:15:A3:00:5B:49 Connected: yes
[UM25C]# [CHG] Device 00:15:A3:00:5B:49 ServicesResolved: yes
[UM25C]# Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[UM25C]# hci0 00:15:A3:00:5B:49 type BR/EDR disconnected with reason 2
[UM25C]# [CHG] Device 00:15:A3:00:5B:49 ServicesResolved: no
[CHG] Device 00:15:A3:00:5B:49 Connected: no
```

kolinger, 2024-05-06:
Hi,
this is not easy to debug since the error message is generic. The underlying bluetooth layer just gives up for whatever reason and doesn't tell why.

You are saying that official app on phone barely works and that's not something I would expect if multiple computers with different operating systems don't. What barely means? Do you see data from the meter like voltage?

Bluetooth has funny bahaviour where the communication is plain broken yet discovery works so it seem to "connect" and then fails so until you see actual data it doesn't mean anything.

If meter didn't report data anywhere I would start to think there is problem with the meter. But that doesn't make sense if Android app sees data.

Anyway in past there were multiple instances where people had issue with construction of the UMxxC meters. The meter is sandwich of two boards where one is the main doing everything and the other is handling bluetooth and they communicate via spring contacts and this spring contact connection is sometimes unreliable. Disassembling and reassembling the meter and/or cleaning the spring contacts and/or landing pads did help people to solve this issue. But of course this only makes sense if even Android app can't see the data.

AlexHuebi, 2024-05-06:
Hi, 
I have to correct myself: the "barely works" was kinda wrong. 

The Connection between the phone and USB meter works and I am getting data points that correlate with the ones I am seeing on the display. But I have been unable to export the recorded data and the UI is not really great to analyze it when I charged something big like a power bank. 

But sometimes, when the session is longer, the data points seem to stop but that may be a different problem from the app which also was one of the reasons why I want to get away from it. 

As far as I could tell, the built in BT-UART Transceiver is the `dx-bt18` which I'd guess, didn't change over the different revisions. Sadly though, I was unable to find any firmware versions that'd explain any difference in behavior.

kolinger, 2024-05-06:
Well unless they changed the protocol completely (like switching from Bluetooth to Bluetooth LE) then I don't see why the meter doesn't return any Bluetooth services. This step is universal and even if those services changed there should be something not nothing.

What methods of connection did you try? Did you try to pair first to create serial/COM ports?

If you didn't try to use UM25C Serial then give it a try. Try to pair with the meter - can be done normally via Settings on Windows or via `rfcomm bind 0 aa:bb:cc:dd:ee:ff` on Linux. This should generate couple of COMx on Windows or /dev/rfcomm0 on Linux and give this to RD-USB via the UM25C Serial profile and see if something happens. More info in readme.

AlexHuebi, 2024-05-06:
I tried that just now and it worked without any issues. 

So it seems like the issue isn't the bluetooth connection itself but maybe the way how the python script tries to connect?

kolinger, 2024-05-06:
Python side is very plain there is not much to do, all is handled by the library and (mostly) underlying bluetooth stack of your OS. If find_services returns empty list then python can't do anything about it.

Did you pair the meter with your OS before trying RD-USB for the first time (with the regular UM25C direct connection)?

Does all the platforms you tested (Windows, Linux) behave exactly the same (Bluetooth service not found)?

AlexHuebi, 2024-05-06:
The error in RD-USB was both in Windows and Linux the same. 

I also tried both first pairing it and not pairing the device and both still had the same error message.

kolinger, 2024-05-06:
From you message it's not clear if you tested before pairing for the first time or try to unpair and then test. Did you make test where your OS isn't paired and doesn't have previous paired profile saved?

This is very strange. I have no explanation for such situation where everything seems to be correct yet no services are found, such situation shouldn't be possible.

Did you try other PC software? For example sigrok supports UM series over Bluetooth and there are other projects and scripts (rdumtool) that can sample UM25C over Bluetooth, this way you cold confirm if indeed everything else is working and just RD-USB doesn't.

AlexHuebi, 2024-05-06:
Hi, 
sorry for the grammatical mess that I wrote. 
Yes, I initially tried to use it without pairing the device to the PC. Because I got the strange error, I then tried to pair it with my device where it connected for a few seconds, then disconnected again. (debug log from that is in the initial message)
I then deleted the device, and tried once more then on all the mentioned other devices with the same steps. 
Unpaired (not working) -> paired (not working and disconnected after a few seconds) -> deleted and tried one last time (not working) 

As for the other software: I will try sigrok and rdumtool and get back to you with an update.

monojk, 2024-10-05:
I have the same error (see picture) although I could connect (before calling rd-usb) to bluetooth via the Bluetooth icon (see picture, UMC34C is not connected showing red x). Setup could identify the UMC34C as the shown MAC address is correct.
[image]

kolinger, 2024-10-05:
I have no idea what it means when the Raspbian shows the device in the dropdown like that. That means it knows about it? It means it's paired? What it means? Regardless what it means try to get rid of it. Remove the UM34C Bluetooth profile form the OS or whatever you need to do so it doesn't even show it knows about the UM34C.

If its not possible or it doesn't work anyway then follow [Bluetooth issues](https://github.com/kolinger/rd-usb?tab=readme-ov-file#usage) in the readme.

monojk, 2024-10-05:
I removed the device from the Raspbian bluetooth dropdown. Then
```
pi@rp4b:~/rd-usb $ hcitool scan
Scanning ...

	00:15:A3:00:59:95	n/a
pi@rp4b:~/rd-usb $ 
pi@rp4b:~/rd-usb $ sudo rfcomm bind 0 00:15:A3:00:59:95
pi@rp4b:~sudo /usr/local/bin/python3.11 web.py
```
Problem persists

kolinger, 2024-10-05:
If you are using the "UM35C Serial" option with rfcomm then there is not way that you get the same error since the error is specific to direct Bluetooth connect (the "UM35C" option). Thus are you sure you have correct "UM35C Serial" option selected? Or what error are you seeing now?

monojk, 2024-10-05:
Ah, I had to select "UM34C" serial from the pulldown, not clicking the setup link as I did before. Now it works. Thanks!

kolinger, 2024-11-18:
I tried to reproduce the issue with latest Raspbian and it works regardless what I try, you don't need to pair with the UMxxC to use rd-usb but if you do it works as well. I cannot reproduce this issue where I would get the:

> Bluetooth service not found, try to initiate Setup again

Thus I have no idea why some machines don't work with the direct Bluetooth connection. Since The UMxxC Serial works then I would suggest using the Serial/COM port/rfcomm method outlined in the readme as alternative, it's bit more complicated to setup but it seems to work universally everywhere.

closed 2024-11-18 by kolinger