kolinger/rd-usb #46: Running the program under Windows 10 triggers Tojan.gen.MBT

Vyeche, 2024-05-12:
Running the program under Windows 10 triggers Tojan.gen.MBT

[image]

Razor512, 2024-05-12:
I have noticed that as well on newer versions when using it on a compute stick device (low power to more efficiently do longer term monitoring). I have had to downgrade because windows defender goes crazy. Hopefully there will be a way to get it to coexist with AV software.

kolinger, 2024-05-12:
I can't do much about that unfortunately. I don't know why it's detected as virus, I don't know what Norton or Defender doesn't like about it. If you know what or why then please tell me I will try to work around it but since I don't know then I can only hope that the antivirus vendor fixes this.

https://www.virustotal.com/gui/file/7ffae410b8b6b9901b48b9ace3c1c5c808a44559ae80d94ba03bc95ac9a4a06c
7 of 70 do think it's virus of some kind, not sure why

This project didn't change much since the beginning, there were no major changes in the way the app works or what it uses for libraries/technologies so there is no reason why older version isn't flagged the same as the newer version.

If you want workaround then you can make exception in antivirus to make it shut-up about it. If that isn't possible or you don't want to bypass your antivirus then you can to run python manually. Don't use .exe executable and instead install python as per readme "Installation - Source code", it's more complicated and more work, that's why the .exe versions exist but the python-way will work the same once setup is completed. The most likely thing the antiviruses don't like is the way the .exe executable is created from python but I can't change that, I would need to rewrite the project to another language to make it different, not realistic.

Razor512, 2024-05-12:
So far, I have largely been trying to get windows defender to give me the option to submit files in the hopes that they can test it and find a more universal solution. One issue is even if you whitelist the exe, while it is running the AV will still eventually kill the process after it has been running for a while when one of the behavioral aspects goes crazy, unless you roll back to an older version. Overall windows defender will kill it after a few minutes to a few hours of use.

Older logs from my compute stick device before I rolled it back to version 1.19, as restoring the file repeatedly and setting the "threat" to allowed, was not working, and just needed reliable data capture.

Here are the windows defender log files from that time: 
[detection logs.txt](https://github.com/kolinger/rd-usb/files/15287972/detection.logs.txt)

The app also logged when when the AV went crazy as well. (github does not support the file type unless zipped, though one excerpt from it would look like this:

2024-04-07 22:52:11,400 - ERROR - <class 'OSError'>
Traceback (most recent call last):
  File "webapp\backend.py", line 359, in retry
  File "interfaces\wrapper.py", line 32, in connect
  File "interfaces\wrapper.py", line 25, in run
  File "multiprocessing\process.py", line 121, in start
  File "multiprocessing\context.py", line 224, in _Popen
  File "multiprocessing\context.py", line 336, in _Popen
  File "PyInstaller\hooks\rthooks\pyi_rth_multiprocessing.py", line 75, in __init__
  File "multiprocessing\popen_spawn_win32.py", line 74, in __init__
OSError: [WinError 225] Operation did not complete successfully because the file contains a virus or potentially unwanted software

Other errors that will happen when windows defender goes crazy, though unsure if linked or not, is:

2024-04-06 15:56:31,972 - ERROR - <class 'sqlite3.OperationalError'>
Traceback (most recent call last):
  File "webapp\backend.py", line 270, in run
  File "webapp\backend.py", line 403, in log
  File "webapp\backend.py", line 396, in emit
  File "utils\storage.py", line 210, in log
sqlite3.OperationalError: database is locked

[error.log.zip](https://github.com/kolinger/rd-usb/files/15287991/error.log.zip)

kolinger, 2024-05-13:
The first stack trace is when RD-USB is launching extra process for communication, the extra process is RD-USB itself, thus this doesn't make sense, if Defender allows to launch the app in the first place why then says that another copy is virus when first copy is running already? That's too late if it was virus it would already do damage. Spawning processes is for sure "suspicious" but this is exactly what 1.19 does as well. This extra process is used as workaround for better reliability since on Windows it's not uncommon to see situation where IO/Bluetooth call just freeze, that's why RD-USB is launching these calls as external process so the process can be killed and restarted when Windows refuses to cooperate. [This was added on the beginning of the project 5 years ago](https://github.com/kolinger/rd-usb/releases/tag/1.0.0-beta.4). The other stack trace seems unrelated. It just says database is already used by some other process perhaps as sideffect when defender denies all file system access.

This make sense why behavior detection is triggered but there is no reason why it should block newer version and not older. The thing that get's newer version blocked is what the old version does the same.

The only thing I see as difference between old and new version is age. 1.18+1.19 did have more users in it's lifetime than newer versions but the behavior from antivirus standpoint is the same. Thus I can only assume that what RD-USB does is suspicious by default and if the specific executable doesn't have build up reputation and seems unknown then it gets blocked because why not.

I did check what did change from 1.19.0 to 1.22.1 and I don't see anything suspicious that was added or changed from behavior standpoint. That's why 1.19.0 should be blocked as well since it's doing the same suspicious spawning of it self.

Unfortunately this is what program with low user count and thus no reputation can expect and there is nothing reasonable we can do about it. Other programs do the same suspicious things and are allowed just because there are more popular and thus have bigger reputation. Thus they are unlikely to be viruses I guess.

In this example running python directly should help since then python spawn itself and python has enough of reputation for it's executable to avoid this nastyness from the antiviruses.

If your antivirus doesn't honor what you tell it to do and terminates processes eventually even when you tell it to ignore it then it's just bug. If you whitelist the exe then antivirus shouldn't block it anyway. That's why there is white-list. I would don't mind if the antivirus thinks it's virus even when it's not but then it should allow me to whitelist whatever it doesn't like. That's how it normally works. If antivirus blocks it anyway even if told not to do  then it's just broken and I would simply replace it with some other antivirus that isn't broken as much.

If older version works for you and does what you want then be happy with whatever work. The newer version don't add anything revolutionary and terms of output or UI.

kolinger, 2024-07-15:
Closing since I can't do anything about it nor I don't know why it happens.

closed 2024-07-15 by kolinger