kolinger/rd-usb #47: Please find a better alternative to pybluez

blshkv, 2024-06-06:
This project is dead officially, see the following note on their website:
```
This project is not under development
Consider some alternatives:

[Bleak](https://github.com/hbldh/bleak): Cross-platform Bluetooth Low Energy client library
[Bless](https://github.com/kevincar/bless): Cross-platform Bluetooth Low Energy server library
[PyQt](https://www.riverbankcomputing.com/software/pyqt/): Python bindings for cross-platform [QtBluetooth](https://www.riverbankcomputing.com/static/Docs/PyQt6/api/qtbluetooth/qtbluetooth-module.html) API
```

rd-usb relay on bleak already, and pyQt is also required indirectly (via pywebview).

Please consider switching to a better supported alternative.

kolinger, 2024-06-06:
Hi,
I know pybluez isn't in active development and I would happily use something else but it's not easy to find alternative. I did already tried to find alternative but I didn't find any.

For example the 3 mentioned projects:

1) Bleak - rd-usb is using bleak already for LE but that's not a option for regular Bluetooth (RFCOMM) since bleak is LE-only.
2) Bless - is also LE-only, bleak is already used for LE so there is no need for Bless
3) PyQt looks like it can do the job but for heavy cost, since it's large dependency `The bindings are implemented as a set of Python modules and contain over 1,000 classes`. We just need RFCOMM not whole Qt binding or the whole Qt - is there option to use small fraction of PyQt? I don't think that's possible. The pywebview is used only for GUI version and Qt frontend isn't used and the headless version version doesn't use pywebview at all so we would need to create dependency on Qt just for RFCOMM and that doesn't sound like great solution since it would multiple the size of this project for one function. Correct me if I'm wrong about this - I don't know PyQt or Qt well enough so I'm not sure if there is way to use just Bluetooth part and not the whole thing.

pybluez is used only for RFCOMM nothing else.

Are there any other regular Bluetooth cross-platform libraries with RFCOMM support? I did try to find those some time ago so I'm not sure if the situation didn't change. Out of those three options only PyQt could apply at all and not very well.

blshkv, 2024-06-06:
Thanks for sharing the details. Unfortunately, I'm not an expert in this aria and can't recommend anything.
I hit a https://github.com/airgproducts/pybluez2 fork by an accident, but there are no updates since 2022 either...

kolinger, 2024-06-06:
PyBluez2 doesn't install from pypi either (last eggs are for Python 3.10) and that's the only problem with PyBluez - it works fine but it also isn't possible to install from pypi since it's missing the eggs for more modern Python (last eggs are for 3.7). RD-USB targets Python 3.11 currently.

I don't see much motivation to switch to PyQt if PyBluez works since you will exchange one drawback for another one, if PyBluez stops working then fixing PyBluez or switching to PyQt is the only option I suppose.

You have issue with PyBluez? It doesn't compile on your system?

blshkv, 2024-06-06:
the latest official release (0.23) is broken, but the git compiles fine (and that is inline with requirments.txt).
Python 3.12 could be a problem, since Gentoo (a distro of my choice) switched to this version.

I pushed the latest snapshot, but haven't tested it:
https://github.com/pentoo/pentoo-overlay/commit/2a02af493b7f6503b1ad1249a48f88abf30934bd

kolinger, 2024-06-06:
I can confirm pybluez compiles with Python 3.12.3 as well. I do believe that if build is broken with future versions then there is good chance someone from community will help since there is no true alternative. This already happened when [Python 3.11 required adjustments](https://github.com/pybluez/pybluez/pull/410).

Thus I would classify pybluez as the best option for RFCOMM for the near future. If there was major issue then I would investigate the PyQt way or other alternative if any come about by then.

What do you think about this route?

blshkv, 2024-06-06:
ok, great. Let's not break something which is not really broken ;-)

Thanks for the great software btw! I got tc66c recently and rf-usb is a lifesaver!

closed 2024-06-06 by blshkv