kolinger/rd-usb #49: Can't install via pip

Inigma77, 2024-06-25:
I may be doing something wrong but i just can't seem to get it to install via pip. It throws this error every time: 
[image]
There is probably a simple solution that I'm just not seeing. I'm running python 3.12 and have the fix for pendulum. I can provide any more info that could help

kolinger, 2024-06-25:
The build failed for some reason but I don't see why, the snippet you show only says it failed. There is no more information?

I tried Python 3.12 on Windows and requirements_headless_new.txt install everything fine.

Perhaps you don't have required dependencies for building/compiling of C/C++? You should see more specific description of why it failed, like it's missing Microsoft Visual C++ or something like that...

Inigma77, 2024-06-25:
I installed Visual C++ and tried again but got the same issue, I very well could be missing some dependencies or something. This is the bottom half with the error type
[image]

Inigma77, 2024-06-25:
I just did it again and got different errors this time
[image]

kolinger, 2024-06-25:
> This is the bottom half with the error type

The part you posted doesn't look like missing dependencies - the setuptools internally failed, that's unusual... That's the whole output - nothing lead up to this?

> I just did it again and got different errors this time

Network connection issue to the github servers or github server have issues - not sure - not related to python/pybluez/rd-usb.

Inigma77, 2024-06-25:
There's the full output of what happens when I try to install it

```
C:\Users\Johnny\Downloads\rd-usb-source-v1.22.4>C:\Users\Johnny\AppData\Local\Programs\Python\Python312\python.exe -m pip install -r requirements_headless_new.txt
Collecting pybluez (from -r requirements_headless_new.txt (line 9))
  Cloning https://github.com/pybluez/pybluez.git to c:\users\johnny\appdata\local\temp\pip-install-kaisyda9\pybluez_2ea24c8e8a81447c91cc421babbda76c
  Running command git clone --filter=blob:none --quiet https://github.com/pybluez/pybluez.git 'C:\Users\Johnny\AppData\Local\Temp\pip-install-kaisyda9\pybluez_2ea24c8e8a81447c91cc421babbda76c'
  Resolved https://github.com/pybluez/pybluez.git to commit 82cbba8a1ebd4c1e3442dfafd8581d58c50fa39e
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [58 lines of output]
      running dist_info
      creating C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info
      writing C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\PKG-INFO
      writing dependency_links to C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\dependency_links.txt
      writing eager_resources to C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\eager_resources.txt
      writing requirements to C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\requires.txt
      writing top-level names to C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\top_level.txt
      writing manifest file 'C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\SOURCES.txt'
      reading manifest file 'C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      warning: no files found matching 'README'
      warning: no files found matching '*.c' under directory 'macos'
      warning: no previously-included files matching '*.pyc' found under directory '*'
      warning: no previously-included files matching '*.o' found under directory '*'
      warning: no previously-included files matching '*.opp' found under directory '*'
      adding license file 'COPYING'
      writing manifest file 'C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez.egg-info\SOURCES.txt'
      creating 'C:\Users\Johnny\AppData\Local\Temp\pip-modern-metadata-lrc_dyt_\PyBluez-0.30.dist-info'
      Traceback (most recent call last):
        File "C:\Users\Johnny\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Users\Johnny\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Johnny\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\build_meta.py", line 368, in prepare_metadata_for_build_wheel
          self.run_setup()
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\build_meta.py", line 497, in run_setup
          super().run_setup(setup_script=setup_script)
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\build_meta.py", line 313, in run_setup
          exec(code, locals())
        File "<string>", line 84, in <module>
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 184, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 200, in run_commands
          dist.run_commands()
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\dist.py", line 976, in run_command
          super().run_command(command)
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\command\dist_info.py", line 97, in run
[... cut ...]
        File "C:\Users\Johnny\AppData\Local\Temp\pip-build-env-383xcez3\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 862, in get_command_obj
          cmd_obj = self.command_obj[command] = klass(self)
                                                ^^^^^^^^^^^
      TypeError: 'NoneType' object is not callable
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

kolinger, 2024-06-25:
I see, this is caused by recent pip version, specifically 24.x. It did work for me since I had 23.x. The Python 3.12.1 did have 23.x, the Python 3.12.4 has 24.0, I also tried 24.1 but that doesn't work either, so 23.x is the latest working.

Thus as workaround downgrade the pip to latest 23.x and it should work:

```
C:\Users\Johnny\AppData\Local\Programs\Python\Python312\python.exe -m pip install --upgrade pip==23.3.2
```

This should be eventually fixed in the pybluez library I suppose, not sure what is the cause...

Inigma77, 2024-06-25:
The downgrade has made it work for me. Thanks for the help and thanks for the program :)

Inigma77, 2024-06-25:
Downgraded pip, its getting further this time but still erroring out, looks more like the reported bugs. Currently installing what i think is the right thing to fix the c builder but not sure if anything else looks wrong
```
C:\Users\Johnny\Downloads\rd-usb-source-v1.22.4>C:\Users\Johnny\AppData\Local\Programs\Python\Python312\python.exe -m pip install -r requirements_headless_new.txt
Collecting pybluez (from -r requirements_headless_new.txt (line 9))
  Cloning https://github.com/pybluez/pybluez.git to c:\users\johnny\appdata\local\temp\pip-install-_go185uu\pybluez_33e35a2e6ba7465d9fdb5596aa3a0d48
  Running command git clone --filter=blob:none --quiet https://github.com/pybluez/pybluez.git 'C:\Users\Johnny\AppData\Local\Temp\pip-install-_go185uu\pybluez_33e35a2e6ba7465d9fdb5596aa3a0d48'
  Resolved https://github.com/pybluez/pybluez.git to commit 82cbba8a1ebd4c1e3442dfafd8581d58c50fa39e
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting Flask~=2.2.5 (from -r requirements_headless_new.txt (line 1))
  Using cached Flask-2.2.5-py3-none-any.whl.metadata (3.9 kB)
Collecting pyserial~=3.5 (from -r requirements_headless_new.txt (line 2))
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting python-socketio~=5.5.2 (from -r requirements_headless_new.txt (line 3))
  Using cached python_socketio-5.5.2-py3-none-any.whl.metadata (3.2 kB)
Collecting python-engineio~=4.3.1 (from -r requirements_headless_new.txt (line 4))
  Using cached python_engineio-4.3.4-py3-none-any.whl.metadata (2.2 kB)
Collecting appdirs~=1.4.4 (from -r requirements_headless_new.txt (line 5))
  Using cached appdirs-1.4.4-py2.py3-none-any.whl.metadata (9.0 kB)
Collecting bleak~=0.21.1 (from -r requirements_headless_new.txt (line 7))
  Using cached bleak-0.21.1-py3-none-any.whl.metadata (5.3 kB)
Collecting pycryptodome~=3.20.0 (from -r requirements_headless_new.txt (line 8))
  Using cached pycryptodome-3.20.0-cp35-abi3-win_amd64.whl.metadata (3.4 kB)
Collecting pendulum~=3.0.0 (from -r requirements_headless_new.txt (line 13))
  Using cached pendulum-3.0.0-cp312-none-win_amd64.whl.metadata (7.0 kB)
Collecting Werkzeug>=2.2.2 (from Flask~=2.2.5->-r requirements_headless_new.txt (line 1))
  Using cached werkzeug-3.0.3-py3-none-any.whl.metadata (3.7 kB)
Collecting Jinja2>=3.0 (from Flask~=2.2.5->-r requirements_headless_new.txt (line 1))
  Using cached jinja2-3.1.4-py3-none-any.whl.metadata (2.6 kB)
Collecting itsdangerous>=2.0 (from Flask~=2.2.5->-r requirements_headless_new.txt (line 1))
  Using cached itsdangerous-2.2.0-py3-none-any.whl.metadata (1.9 kB)
Collecting click>=8.0 (from Flask~=2.2.5->-r requirements_headless_new.txt (line 1))
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting bidict>=0.21.0 (from python-socketio~=5.5.2->-r requirements_headless_new.txt (line 3))
  Using cached bidict-0.23.1-py3-none-any.whl.metadata (8.7 kB)
Collecting winrt-Windows.Devices.Bluetooth==2.0.0b1 (from bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_Windows.Devices.Bluetooth-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (1.6 kB)
Collecting winrt-Windows.Devices.Bluetooth.Advertisement==2.0.0b1 (from bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_Windows.Devices.Bluetooth.Advertisement-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (1.3 kB)
Collecting winrt-Windows.Devices.Bluetooth.GenericAttributeProfile==2.0.0b1 (from bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_Windows.Devices.Bluetooth.GenericAttributeProfile-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (1.4 kB)
Collecting winrt-Windows.Devices.Enumeration==2.0.0b1 (from bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_Windows.Devices.Enumeration-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (1.5 kB)
Collecting winrt-Windows.Foundation==2.0.0b1 (from bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_Windows.Foundation-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (992 bytes)
Collecting winrt-Windows.Foundation.Collections==2.0.0b1 (from bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_Windows.Foundation.Collections-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (1.0 kB)
Collecting winrt-Windows.Storage.Streams==2.0.0b1 (from bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_Windows.Storage.Streams-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (1.2 kB)
Collecting winrt-runtime==2.0.0-beta.1 (from winrt-Windows.Devices.Bluetooth==2.0.0b1->bleak~=0.21.1->-r requirements_headless_new.txt (line 7))
  Using cached winrt_runtime-2.0.0b1-cp312-cp312-win_amd64.whl.metadata (676 bytes)
Collecting python-dateutil>=2.6 (from pendulum~=3.0.0->-r requirements_headless_new.txt (line 13))
  Using cached python_dateutil-2.9.0.post0-py2.py3-none-any.whl.metadata (8.4 kB)
Collecting tzdata>=2020.1 (from pendulum~=3.0.0->-r requirements_headless_new.txt (line 13))
  Using cached tzdata-2024.1-py2.py3-none-any.whl.metadata (1.4 kB)
Collecting time-machine>=2.6.0 (from pendulum~=3.0.0->-r requirements_headless_new.txt (line 13))
  Using cached time_machine-2.14.1-cp312-cp312-win_amd64.whl.metadata (21 kB)
Collecting colorama (from click>=8.0->Flask~=2.2.5->-r requirements_headless_new.txt (line 1))
  Using cached colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
[... cut ...]
      running build_ext
      building 'bluetooth._msbt' extension
      error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pybluez
Failed to build pybluez
ERROR: Could not build wheels for pybluez, which is required to install pyproject.toml-based projects

```

Inigma77, 2024-06-25:
Once the build tools got installed it work and installed, thanks for the help :)

closed 2024-06-25 by Inigma77