kolinger/rd-usb #50: Struggling to install via pip on rsapian

Inigma77, 2024-06-27:
I'm back again the time on raspian. I have done tried both install types for pendulum but the each give an error like this. Again its probably a simple fix but i just can't see it.

```
(.venv) --------@raspberrypi:~/Desktop/Demo DIsplay Stuff/rd-usb-1.22.4 $ python3 -m pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pybluez
  Cloning https://github.com/pybluez/pybluez.git to /tmp/pip-install-eufg76ap/pybluez_1088ba9889c9466a8d8a716d31911dfc
  Running command git clone --filter=blob:none --quiet https://github.com/pybluez/pybluez.git /tmp/pip-install-eufg76ap/pybluez_1088ba9889c9466a8d8a716d31911dfc
  Resolved https://github.com/pybluez/pybluez.git to commit 82cbba8a1ebd4c1e3442dfafd8581d58c50fa39e
  Preparing metadata (setup.py) ... done
Collecting Flask~=2.2.5
  Using cached https://www.piwheels.org/simple/flask/Flask-2.2.5-py3-none-any.whl (101 kB)
Requirement already satisfied: pyserial~=3.5 in ./.venv/lib/python3.11/site-packages (from -r requirements.txt (line 2)) (3.5)
Collecting python-socketio~=5.5.2
  Using cached https://www.piwheels.org/simple/python-socketio/python_socketio-5.5.2-py3-none-any.whl (56 kB)
Requirement already satisfied: python-engineio~=4.3.1 in ./.venv/lib/python3.11/site-packages (from -r requirements.txt (line 4)) (4.3.4)
Requirement already satisfied: appdirs~=1.4.4 in ./.venv/lib/python3.11/site-packages (from -r requirements.txt (line 5)) (1.4.4)
Collecting bleak~=0.21.1
  Using cached https://www.piwheels.org/simple/bleak/bleak-0.21.1-py3-none-any.whl (137 kB)
Requirement already satisfied: pycryptodome~=3.20.0 in ./.venv/lib/python3.11/site-packages (from -r requirements.txt (line 8)) (3.20.0)
Collecting pendulum~=2.1.2
  Using cached https://www.piwheels.org/simple/pendulum/pendulum-2.1.2-cp311-cp311-manylinux_2_36_armv7l.whl (156 kB)
Requirement already satisfied: pywebview~=3.7.0 in ./.venv/lib/python3.11/site-packages (from -r requirements.txt (line 16)) (3.7.2)
Collecting pythonnet~=3.0.3
  Using cached pythonnet-3.0.3-py3-none-any.whl (290 kB)
Requirement already satisfied: screeninfo~=0.8.1 in ./.venv/lib/python3.11/site-packages (from -r requirements.txt (line 18)) (0.8.1)
Collecting Werkzeug>=2.2.2
  Using cached https://www.piwheels.org/simple/werkzeug/werkzeug-3.0.3-py3-none-any.whl (227 kB)
Collecting Jinja2>=3.0
  Using cached https://www.piwheels.org/simple/jinja2/jinja2-3.1.4-py3-none-any.whl (133 kB)
Collecting itsdangerous>=2.0
  Using cached https://www.piwheels.org/simple/itsdangerous/itsdangerous-2.2.0-py3-none-any.whl (16 kB)
Collecting click>=8.0
  Using cached https://www.piwheels.org/simple/click/click-8.1.7-py3-none-any.whl (97 kB)
Collecting bidict>=0.21.0
  Using cached https://www.piwheels.org/simple/bidict/bidict-0.23.1-py3-none-any.whl (32 kB)
Collecting dbus-fast<3,>=1.83.0
  Using cached https://www.piwheels.org/simple/dbus-fast/dbus_fast-2.22.1-cp311-cp311-manylinux_2_36_armv7l.whl (3.6 MB)
Requirement already satisfied: typing-extensions>=4.7.0 in ./.venv/lib/python3.11/site-packages (from bleak~=0.21.1->-r requirements.txt (line 7)) (4.12.2)
Requirement already satisfied: python-dateutil<3.0,>=2.6 in ./.venv/lib/python3.11/site-packages (from pendulum~=2.1.2->-r requirements.txt (line 13)) (2.9.0.post0)
Requirement already satisfied: pytzdata>=2020.1 in ./.venv/lib/python3.11/site-packages (from pendulum~=2.1.2->-r requirements.txt (line 13)) (2020.1)
Requirement already satisfied: proxy-tools in ./.venv/lib/python3.11/site-packages (from pywebview~=3.7.0->-r requirements.txt (line 16)) (0.1.0)
Collecting clr-loader<0.3.0,>=0.2.6
  Using cached clr_loader-0.2.6-py3-none-any.whl (51 kB)
Collecting cffi>=1.13
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.16.0-cp311-cp311-linux_armv7l.whl (382 kB)
Collecting MarkupSafe>=2.0
  Using cached https://www.piwheels.org/simple/markupsafe/MarkupSafe-2.1.5-cp311-cp311-linux_armv7l.whl (26 kB)
Requirement already satisfied: six>=1.5 in ./.venv/lib/python3.11/site-packages (from python-dateutil<3.0,>=2.6->pendulum~=2.1.2->-r requirements.txt (line 13)) (1.16.0)
Requirement already satisfied: pycparser in ./.venv/lib/python3.11/site-packages (from cffi>=1.13->clr-loader<0.3.0,>=0.2.6->pythonnet~=3.0.3->-r requirements.txt (line 17)) (2.22)
Building wheels for collected packages: pybluez
  Building wheel for pybluez (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/msbt.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/bluez.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/macos.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/ble.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/__init__.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/btcommon.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      running build_ext
      building 'bluetooth._bluetooth' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/bluez
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC "-I/home/-----/Desktop/Demo DIsplay Stuff/rd-usb-1.22.4/.venv/include" -I/usr/include/python3.11 -c bluez/btmodule.c -o build/temp.linux-aarch64-cpython-311/bluez/btmodule.o
      In file included from bluez/btmodule.c:21:
      bluez/btmodule.h:5:10: fatal error: bluetooth/bluetooth.h: No such file or directory
          5 | #include <bluetooth/bluetooth.h>
            |          ^~~~~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pybluez
  Running setup.py clean for pybluez
Failed to build pybluez
Installing collected packages: pybluez, MarkupSafe, itsdangerous, dbus-fast, click, cffi, bidict, Werkzeug, python-socketio, pendulum, Jinja2, clr-loader, bleak, pythonnet, Flask
  Running setup.py install for pybluez ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for pybluez did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      running install
      /home/-----/Desktop/Demo DIsplay Stuff/rd-usb-1.22.4/.venv/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/msbt.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/bluez.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/macos.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/ble.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/__init__.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/btcommon.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      running build_ext
      building 'bluetooth._bluetooth' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/bluez
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC "-I/home/------/Desktop/Demo DIsplay Stuff/rd-usb-1.22.4/.venv/include" -I/usr/include/python3.11 -c bluez/btmodule.c -o build/temp.linux-aarch64-cpython-311/bluez/btmodule.o
      In file included from bluez/btmodule.c:21:
      bluez/btmodule.h:5:10: fatal error: bluetooth/bluetooth.h: No such file or directory
          5 | #include <bluetooth/bluetooth.h>
            |          ^~~~~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

```

kolinger, 2024-06-27:
Try

```
apt-get install libbluetooth-dev
```

Inigma77, 2024-06-27:
That fixed it sorry for all the bothering and thank you again:)

closed 2024-06-27 by Inigma77