kolinger/rd-usb #54: Rates below 1sec not available

Kabron287, 2025-02-18:
Despite of settings, rates: 100ms, 250ms, 500ms works at 1s interwal.

kolinger, 2025-02-18:
Depending on the connection/meter model some of the faster rates may not make sense, since the measurement time may be too long/longer than the measuring interval, thus the measurement time will cap the interval at specific period. Some of the models can reply faster than others. What model you are using and with what type of connection? The meters aren't made for high sample rate so I won't be surprised that the limitation would be meter but I can check where the code is spending most of the time if you tell me details.

Kabron287, 2025-02-18:
I use UM25C

kolinger, 2025-02-18:
TL;DR
All *polling intervals will work* (with specific meter/connection) but the **100ms and 250ms is useless** even under right conditions since it seems that the meters internally sample at 500ms, thus anything faster than that will only generate duplicate measurements.

---

There are two different rates - the polling rate of the PC and the sample rate of the meter. The application can only set the polling rate.

I tested with UM35 and the result was 3 polls per second, yet there was quite a lot repetition, so it seems that at 3 polls per second the polling interval is faster than whatever internal sampling rate the meter has.

The TC66C over USB polls 10 measurements per second just fine although again the measurements are repeating way too often so it seems like the internal sampling/averaging measurement of the meters is always slower than how fast the meters can reply to polling.

Thus the 100ms/250ms interval is misleading since some meters can reply as fast but it's pointless since the extra samples are duplicates. Thus these meters for sure aren't made for high sample rate - even if the communication is fast enough - the internal sampling will cap the sample rate at 2 measurements per second by eyeballing it.

This would match the "refresh rate" of display with value of 2 Hz that is specified by manufacturer - thus the sample rate seems to be tied to the display update rate (and that's a shame) and thus anything faster than 500ms isn't of any use even if it's possible.

---

Thus we could happy remove the 100ms/250ms option, what do you think?

Kabron287, 2025-02-18:
Leave as is.

closed 2025-02-18 by kolinger